ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA #1: How to run those python Codes to generate verilog codes step by step?

pualdelis, 2018-07-26:
Hi, I'm interested in your projects, but I don't konw how to use those python codes, any instuctions? thanks

ZFTurbo, 2018-07-26:
If you need just veilog to exactly recreate device, then you can check verilog folder it has everything you need. In case you want to generate everything from scratch look at the "How to run" section. There are 5 scripts you need to run one by one. You must have "Python 3.5, Tensorflow >=1.4.0, Keras >=2.1.3" installed.

pualdelis, 2018-08-10:
Thanks a lot. In fact, i found that those python codes generate verilog codes only by write to file and use text replacement. In this case, may i understand like this, the work of python is to train the network and generate a text file of verilog codes by using foregoing parameters, and without really understanding it's meaning?
lastly, have those verilog CNN codes being trained and i can use it directly?
How did you come up with this brilliant idea, is there any book or paper you refered to ?

ZFTurbo, 2018-08-14:
Python code mostly used to obtain weights for neural net, then find how much precision needed for fix point representation. Next generate verilog text file for neural net with calculated precision and required number of convolution blocks. The last part with verilog generation actually written not very flexible currently.

No, there was no paper. We wrote our own paper, but it still under rewiew for publication.

pualdelis, 2018-08-16:
thanks a lot，if your paper is published, please let me know. Thanks again

ZFTurbo, 2018-08-30:
We put preprint on arxiv: https://arxiv.org/abs/1808.09945