ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA #3: Different components configure

mrpossible, 2019-01-07:
Hi, 

I tried to buy the same components as your spec, but it seems that there are some slight differnce and after 100% sucessful build, it didn't work at all. 

Can you help to guide me how to configure with my components here, as I try to plug in as far as I know but the LCD is not working and didn't show anything. 

[image]
[image]

Aobo-xd, 2019-01-07:
您好！
    您的来信已收到，我会尽快查看！
祝好

ZFTurbo, 2019-01-07:
It's different screen. We use SPI interface while yours is some kind of parallel one. Can you send exact model name of your screen?

ZFTurbo, 2019-01-07:
As I remember we used this one: https://ru.aliexpress.com/item/2-8-320-240-SPI-TFT/32847628219.html

mrpossible, 2019-01-07:
this is also SPI, I tried to buy the same but somehow it's differnt
Here is the one I bought on ebay:
https://www.ebay.com/itm/223145284611
FX- 2.8" 240x320 SPI TFT LCD Touch Panel ILI9341 5V 3.3V for Arduino RPi
ESP8266

ZFTurbo, 2019-01-08:
On your foto I see 8 pins for LCD connection, but SPI interface requires only one:
https://en.wikipedia.org/wiki/Serial_Peripheral_Interface

Also some of your wires connected to SD-card pins, which is not used in current project.

mrpossible, 2019-01-09:
Btw, I didn't see the pins XCLK, PCLK from Camero connected to any where in the diagram?
And pin D1 and D3 seems to be flipped? is that correct? 

[image]

ZFTurbo, 2019-01-10:
I will be able to check device connections at Monday. I will post if diagram is totally correct then.

ZFTurbo, 2019-01-14:
We checked connections today. It must be like that:
https://github.com/ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA/blob/master/images/Connection-scheme.png

We updated picture in project.