ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA #4: Pin Planner for FPGA

gabrielchin96, 2019-02-04:
Hi ! 

This is an awesome project! I came across your project and wanted to implement it in FPGA however I'm a little bit unsure on the manual assignments for Pin Planner in the FPGA. Is there a specific pin to be allocated for the node not found in the FPGA datasheet?

Thanks in advance!

Aobo-xd, 2019-02-04:
您好！
    您的来信已收到，我会尽快查看！
祝好

ZFTurbo, 2019-02-05:
Could you please provide additional details, because if you use projects with totally same connections you won't have any problem with Pin Planner. Do you have some kind of error from Quartus? If, yes, please post error message here.

gabrielchin96, 2019-02-07:
The warnings i received in the quartus is as below,
Thanks for this! 

Warning (15536): Implemented PLL "pll_for_disp:pll2|altpll:altpll_component|pll_for_disp_altpll:auto_generated|pll1" as Cyclone IV E PLL type, but with warnings
	Warning (15559): Can't achieve requested value multiplication of 143 for clock output pll_for_disp:pll2|altpll:altpll_component|pll_for_disp_altpll:auto_generated|wire_pll1_clk[0] of parameter multiplication factor -- achieved value of multiplication of 103
	Warning (15559): Can't achieve requested value division of 50 for clock output pll_for_disp:pll2|altpll:altpll_component|pll_for_disp_altpll:auto_generated|wire_pll1_clk[0] of parameter division factor -- achieved value of division of 36
	Info (15099): Implementing clock multiplication of 103, clock division of 36, and phase shift of 0 degrees (0 ps) for pll_for_disp:pll2|altpll:altpll_component|pll_for_disp_altpll:auto_generated|wire_pll1_clk[0] port
Warning (292013): Feature LogicLock is only available with a valid subscription license. You can purchase a software subscription to gain full access to this feature.
Warning (15714): Some pins have incomplete I/O assignments. Refer to the I/O Assignment Warnings report for details
Critical Warning (169085): No exact pin location assignment(s) for 42 pins of 87 total pins. For the list of pins please refer to the I/O Assignment Warnings table in the fitter report.
Warning (176127): The parameters of the PLL pll_for_disp:pll2|altpll:altpll_component|pll_for_disp_altpll:auto_generated|pll1 and the PLL pll:pll_for_sdram_0|altpll:altpll_component|pll_altpll:auto_generated|pll1 do not have the same values - hence these PLLs cannot be merged
	Info (176120): The values of the parameter "M" do not match for the PLL atoms pll:pll_for_sdram_0|altpll:altpll_component|pll_altpll:auto_generated|pll1 and PLL pll_for_disp:pll2|altpll:altpll_component|pll_for_disp_altpll:auto_generated|pll1
		Info (176121): The value of the parameter "M" for the PLL atom pll:pll_for_sdram_0|altpll:altpll_component|pll_altpll:auto_generated|pll1 is 12
		Info (176121): The value of the parameter "M" for the PLL atom pll_for_disp:pll2|altpll:altpll_component|pll_for_disp_altpll:auto_generated|pll1 is 103
	Info (176120): The values of the parameter "N" do not match for the PLL atoms pll:pll_for_sdram_0|altpll:altpll_component|pll_altpll:auto_generated|pll1 and PLL pll_for_disp:pll2|altpll:altpll_component|pll_for_disp_altpll:auto_generated|pll1
		Info (176121): The value of the parameter "N" for the PLL atom pll:pll_for_sdram_0|altpll:altpll_component|pll_altpll:auto_generated|pll1 is 1
		Info (176121): The value of the parameter "N" for the PLL atom pll_for_disp:pll2|altpll:altpll_component|pll_for_disp_altpll:auto_generated|pll1 is 9
	Info (176120): The values of the parameter "LOOP FILTER R" do not match for the PLL atoms pll:pll_for_sdram_0|altpll:altpll_component|pll_altpll:auto_generated|pll1 and PLL pll_for_disp:pll2|altpll:altpll_component|pll_for_disp_altpll:auto_generated|pll1
		Info (176121): The value of the parameter "LOOP FILTER R" for the PLL atom pll:pll_for_sdram_0|altpll:altpll_component|pll_altpll:auto_generated|pll1 is 4000
		Info (176121): The value of the parameter "LOOP FILTER R" for the PLL atom pll_for_disp:pll2|altpll:altpll_component|pll_for_disp_altpll:auto_generated|pll1 is 14000
	Info (176120): The values of the parameter "Min Lock Period" do not match for the PLL atoms pll:pll_for_sdram_0|altpll:altpll_component|pll_altpll:auto_generated|pll1 and PLL pll_for_disp:pll2|altpll:altpll_component|pll_for_disp_altpll:auto_generated|pll1
		Info (176121): The value of the parameter "Min Lock Period" for the PLL atom pll:pll_for_sdram_0|altpll:altpll_component|pll_altpll:auto_generated|pll1 is 18456
		Info (176121): The value of the parameter "Min Lock Period" for the PLL atom pll_for_disp:pll2|altpll:altpll_component|pll_for_disp_altpll:auto_generated|pll1 is 17601
	Info (176120): The values of the parameter "Max Lock Period" do not match for the PLL atoms pll:pll_for_sdram_0|altpll:altpll_component|pll_altpll:auto_generated|pll1 and PLL pll_for_disp:pll2|altpll:altpll_component|pll_for_disp_altpll:auto_generated|pll1
		Info (176121): The value of the parameter "Max Lock Period" for the PLL atom pll:pll_for_sdram_0|altpll:altpll_component|pll_altpll:auto_generated|pll1 is 39996
		Info (176121): The value of the parameter "Max Lock Period" for the PLL atom pll_for_disp:pll2|altpll:altpll_component|pll_for_disp_altpll:auto_generated|pll1 is 22222
Critical Warning (176598): PLL "pll_for_disp:pll2|altpll:altpll_component|pll_for_disp_altpll:auto_generated|pll1" input clock inclk[0] is not fully compensated because it is fed by a remote clock pin "Pin_R8"
Warning (335093): The Timing Analyzer is analyzing 18 combinational loops as latches. For more details, run the Check Timing command in the Timing Analyzer or view the "User-Specified and Inferred Latches" table in the Analysis & Synthesis report.
Warning (332174): Ignored filter at qfit2_legacy_fmain_fitter_flow.tcl(117): *ws_dgrp|dffpipe_ve9:dffpipe7|dffe8a* could not be matched with a clock or keeper or register or port or pin or cell or partition
Warning (332049): Ignored set_false_path at qfit2_legacy_fmain_fitter_flow.tcl(117): Argument <to> is not an object ID
	Info (332050): run_legacy_fitter_flow
[... cut ...]
Warning (332049): Ignored set_false_path at qfit2_legacy_fmain_fitter_flow.tcl(117): Argument <to> is not an object ID
	Info (332050): run_legacy_fitter_flow
Critical Warning (332168): The following clock transfers have no clock uncertainty assignment. For more accurate results, apply clock uncertainty assignments or use the derive_clock_uncertainty command.
	Critical Warning (332169): From TOP:neiroset|nextstep (Rise) to TOP:neiroset|nextstep (Rise) (setup and hold)
	Critical Warning (332169): From TOP:neiroset|nextstep (Rise) to TOP:neiroset|nextstep (Fall) (setup and hold)

ZFTurbo, 2019-02-11:
It doesn't look like problem. Project should work.

gabrielchin96, 2019-02-20:
Hi, is it possible to provide testbench to your neuroset verilog coding?

ZFTurbo, 2019-02-25:
Yes. Testbench added:
https://github.com/ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA/blob/master/verilog/code/testbench.v

gabrielchin96, 2019-03-21:
Thanks for the testbench! regarding the testbench added, what was the original image input in the system?