ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA #5: cam_proj.out.sdc

hooper888, 2019-02-19:
Critical Warning (332148): Timing requirements not met
	Info (11105): For recommendations on closing timing, run Report Timing Closure Recommendations in the TimeQuest Timing Analyzer.
Critical Warning (332148): Timing requirements not met
	Info (11105): For recommendations on closing timing, run Report Timing Closure Recommendations in the TimeQuest Timing Analyzer.
Critical Warning (332148): Timing requirements not met
	Info (11105): For recommendations on closing timing, run Report Timing Closure Recommendations in the TimeQuest Timing Analyzer.
   
Hello, there is no file
cam_proj.out.sdc. And this series of errors appears, please help !!!

Aobo-xd, 2019-02-19:
您好！
    您的来信已收到，我会尽快查看！
祝好

ZFTurbo, 2019-02-19:
It's not errors it's warnings. It must work without this file (we don't have it also).

albertlu2013, 2019-02-19:
Hi ZFTurbo
I get the same Critical Warning. 
The programmer is 100% success. But my board can not work. The ili9341 lcd is all white (only back led light).
I found the "TimeQuest Timing Analyzer" is red color. Is your "TimeQuest" also red ?

Thanks

ZFTurbo, 2019-02-20:
_Timing requirements not met_ - this typically means that frequencу is too high, try to decrease it until this warning is gone. But we also have this warning. TimeQuest Timing Analyzer currently green for us.

Regarding white screen: May be screen connection is incorrect?

I will have access to hardware device at monday. It currently successfully work with files from repo, but I will check log of compilation to see the possible problems.

ZFTurbo, 2019-02-20:
Screen from Quartus. Project is ok with these warnings.

https://raw.githubusercontent.com/ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA/master/images/Q-scr-2019-02.png