ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA #12: How to simulate the generated verilog

atarun116, 2020-04-10:
Hi, 

I don't have the hardware ... but I'm interested to compare speedup of result's to understand the code and get an estimate. Can you please tell me how can I do that ?

thanks,
Tarun

ZFTurbo, 2020-04-10:
You can try to use testbench:

https://github.com/ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA/blob/master/verilog/code/testbench.v

atarun116, 2020-04-10:
Hi I tried to simulate testbench.v using xcelium by using 

xrun -f testbench.v 

I encountered the following error :

xrun: *E,FMUK: The type of the file (storage[777]) could not be determined.
xrun: *E,FILEMIS: Cannot find the provided file storage[777].
xrun: *E,FMUK: The type of the file (=) could not be determined.
xrun: *E,FILEMIS: Cannot find the provided file =.
xrun: *E,FMUK: The type of the file (11'b10010111000;) could not be determined.
xrun: *E,FILEMIS: Cannot find the provided file 11'b10010111000;.
xrun: *E,FMUK: The type of the file (storage[778]) could not be determined.
xrun: *E,FILEMIS: Cannot find the provided file storage[778].
xrun: *E,FMUK: The type of the file (=) could not be determined.
xrun: *E,FILEMIS: Cannot find the provided file =.
xrun: *E,FMUK: The type of the file (11'b10010101000;) could not be determined.
xrun: *E,FILEMIS: Cannot find the provided file 11'b10010101000;.
xrun: *E,FMUK: The type of the file (storage[779]) could not be determined.
xrun: *E,FILEMIS: Cannot find the provided file storage[779]

I'm new in field of electronics ... can you suggest How can I run this. 

thanks

congrats for the great work, 
best wishes

ZFTurbo, 2020-04-10:
Do you have access to ModelSim? If I remember we use it for testing.

atarun116, 2020-04-10:
Hi,

No... I don't have its licence. I only have xcelium. 

thanks

atarun116, 2020-04-10:
I think this is due to not able to access TOP.v file in neuroset. it doesn't include all the .v modules.

thanks

ZFTurbo, 2020-04-10:
As I know ModelSim is avilable for free:
https://fpgasoftware.intel.com/?product=modelsim_ae#tabs-2

I can't help you with xcelium. Probably you need to include all needed .v files in testbench somehow.

atarun116, 2020-04-10:
Thnaks for the modelsim ... I can give a try. 
Also I have more query regarding the input image in testbench.v , from which path it's picking its image.

thanks,
Tarun

ZFTurbo, 2020-04-10:
It's hardcoded in initial block. See "storage" variable

atarun116, 2020-04-10:
Hi,
is there any way we can read the image directly by converting it into some format. 
also, is this conversion of 11 bits or 10 bits is in testbench is also dynamic based on your previous stage results? 

thanks

ZFTurbo, 2020-04-10:
Bit for image is found in "r03_find_optimal_bit_for_weights.py". Also there was some code I used to convert PNG to this format used in testbench. I will try to find it.

atarun116, 2020-04-10:
thanks, That will be good so we can see that the output of the simulation is with respect to which image.

thanks

atarun116, 2020-04-10:
Hi, 

One more doubt, the testbench.v is written for :

https://github.com/ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA/blob/master/verilog/code/neuroset/TOP.v

this file  right ?

thanks

ZFTurbo, 2020-04-10:
Yes, must be for TOP

atarun116, 2020-04-10:
Hi,

thanks for your advice ... with some modification, I'm able to finally simulate :) however, 
previous error in the compilation is because  in neroset/conv_TOP.v file, the parameter conv 3 is not set. 
also while simulation is working and result is always 10 and it throws msgs Overflow in conv !.

why result is everytime 10, is it because everytime same test image is hardcoded into testbench.v file ?  can you please check. 

thanks

atarun116, 2020-04-10:
Hi, 

Congrats, the .py files generating verilog files are written with such precision ... I mean a small change could cause a lot hard to determine errors of the generated files. 

kudos for this great work :) 

thanks

ZFTurbo, 2020-04-10:
I added code to convert image to testbench format:
https://github.com/ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA/blob/master/utils/convert_image_for_testbench.py

atarun116, 2020-04-10:
Hi,

thanks for adding the file. it's really helpful. 
Can we use optimal bit setting +1 extra bit to prevent overflow in neuroset/conv.v file? 

thanks

ZFTurbo, 2020-04-10:
If you change optimal bit you need to regenrate all files with this optimal bit. Including image in test_bench.

atarun116, 2020-04-10:
Hi,  

you already considering this option, look @ file [r04_verilog_generator_grayscale_file.py](https://github.com/ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA/blob/master/r04_verilog_generator_grayscale_file.py)  line 137, please correct me if I'm wrong.

thanks

atarun116, 2020-04-10:
> Hi,
> 
> thanks for your advice ... with some modification, I'm able to finally simulate :) however,
> previous error in the compilation is because in neuroset/conv_TOP.v file, the parameter conv on line 3 is not set.
> also while simulation is working and result is always 10 and it throws msgs Overflow in conv !.
> 
> why result is everytime 10, is it because everytime same test image is hardcoded into testbench.v file ? can you please check.
> 
> thanks
hi, 
is there any particular reason for not setting this parameter num_conv in conv_TOP.v file on line 3 ... Can you please check as this is causing a syntax error while simulation. I fixed this in my local directory but I'm curious about the reason. 

thanks

atarun116, 2020-04-10:
> I added code to convert image to testbench format:
> https://github.com/ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA/blob/master/utils/convert_image_for_testbench.py

hi, 

Can you please tell or add the description of this file in read.md. 
also I noticed that the file set bp = 11 at line 112, why is that ? is this to avoid recomputing optimal bit setting ? 

while running this script ... shows an issue : 
"convert_image_for_testbench.py", line 117, in <module>
    out_path = ROOT_PATH + "verilog/initial_image_bitdepth_{}_img_number_{}_answ_{}.v".format(bp+1, use_image, answers[use_image])
IndexError: list index out of range.

can you please explain the usage of this script, this will be very helpful :) 

thanks

ZFTurbo, 2020-04-11:
You need to replace only image part in current testbench. Generator only generate image description without other testbench stuff.

ZFTurbo, 2020-04-11:
> hi,
> is there any particular reason for not setting this parameter num_conv in conv_TOP.v file on line 3 ... Can you please check as this is causing a syntax error while simulation. I fixed this in my local directory but I'm curious about the reason.

It's probably because this parameter is set in TOP module.
```
module TOP(clk, GO, RESULT, we_database, dp_database, address_p_database, STOP);

parameter num_conv = 1;
```

atarun116, 2020-04-12:
Hi, 

why is train and test data consist of folder 10, in standard MNIST dataset ... we don't consider that class.

thanks,

ZFTurbo, 2020-04-12:
In MNIST dataset each class contain number, but in real world we have images without any number on it. So we add additional class for 'no number' class.
Also image from camera have other resolution. In our case it's 320x240. Test folder contains raw images obtained from device camera.

atarun116, 2020-04-12:
Hi, 

thanks for the above explanation... I was expectating something similar. Can you please tell me what is the usage of r04_verilog_generator_grayscale_file.py and [verilog/code/gray_28x28/grays](https://github.com/ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA/blob/master/verilog/code/gray_28x28/grayscale.v)grayscale.v

thanks

ZFTurbo, 2020-04-12:
As I remember this module converts RGB image from camera to 28x28 size.

atarun116, 2020-04-12:
Do I need to include it while running the simulation ?

thanks

ZFTurbo, 2020-04-12:
I don't think so, because in tesetbench you already use grayscale image 28x28.

atarun116, 2020-04-13:
what is the use of the /verilog/imp folder and /verilog/code/lcd and /verilog/code/synt folders ?

thanks

ZFTurbo, 2020-04-13:
`/verilog/imp` - mostly files generated by Quartus for FIFO and PLL, also some top level project files
`/verilog/code/lcd` - verilog files for LCD-screen
`/verilog/code/synt` - verilog to interact with camera, working with SDRAM, FIFO to input/output data from camera and memory.