ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA #13: error when run python code: r05_verilog_generator_neural_net_structure.py

giathinhlenguyen, 2020-08-17:
Hi I tried to run r05_verilog_generator_neural_net_structure.py  and i got this error. Can you show me how to solve it. Thanks
[image]

ZFTurbo, 2020-08-18:
1) Can you please provide your tensorflow and keras version?
2) Also please write on line before this error `print(model.summary())`

giathinhlenguyen, 2020-08-18:
1. i currently use Python 3.7, tensorflow  2.3.0, keras 2.4.3.
2. I tried and seem notthing was printed on the screen.
[image]

ZFTurbo, 2020-08-18:
This repository contains old code. I believe it won't work with TF 2.* There are too much changes comparing with TF 1.*

Try to install: Tensorflow 1.4.0, Keras 2.1.3 like it said in requirements. Probably using TF 1.15 + keras 2.3.* will be ok too.

cbureriu, 2020-12-09:
`print(model.summary())` returns `None`