ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA #17: About Testbench.v   Result

QiQi-OvO, 2021-04-24:
Why do I generate results that don't match the actual results **Answer**:.

QiQi-OvO, 2021-04-25:
Hi, and I want to know the meaning of variables: _memstartp_,_memstartzap_,_mem,filt_,_matrix_,_lvl_,_slvl_,_num_. thanks a lot

closed 2021-11-11 by QiQi-OvO

kangliyu1, 2022-01-10:
> Why do I generate results that don't match the actual results **Answer**:.

Hello, I also encountered the problem that the simulation does not match the results. Can you give me some pointers on how to solve it? Many thanks

QiQi-OvO, 2022-01-10:
你可以试着调一下时钟频率，我记得时钟频率好像会影响到整个结果。过去的时间比较久，我已经不再做这个方面的研究了。

kangliyu1, 2022-01-10:
> 你可以试着调一下时钟频率，我记得时钟频率好像会影响到整个结果。过去的时间比较久，我已经不再做这个方面的研究了。
> […](#)

好的谢谢您的回复，还有个问题就是想问下您，那您怎么知道自己调完时钟频率之后仿真就会对了，他的testbench里的784个11位的storage是代表的什么数呢？那您知道它自带的那个testbench的storage这个784个11位的数是怎么生成的吗？是使用它的哪个python生成的吗？

kangliyu1, 2022-01-10:
> 你可以试着调一下时钟频率，我记得时钟频率好像会影响到整个结果。过去的时间比较久，我已经不再做这个方面的研究了。
> […](#)

我是想单独仿真验证它的神经网络加速器的代码，但是刚刚在testbench中更改频率（改动了clk）之后发现结果并没有发现变化，如果您有空的话希望可以指点一下，万分感谢前辈