ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA #19: Timing Simulation not working

zee9999, 2021-11-08:
First of all, this project is really good. We ran the functional simulation on ModelSim and worked perfectly however when we ran the timing simulation-after the synthesis on Quartus lite-it kept giving us a RESULT"1111" for a while and no output was given after that. [Here is a link with the timing simulation](https://drive.google.com/file/d/19FcCm_-t_nT62JZyAnLk7dvv7iM7hINw/view?usp=sharing)

ZFTurbo, 2021-11-09:
The verilog code author says it's ok. Can you check the end of waveform?

kangliyu1, 2022-01-10:
> 
@zee9999 @ZFTurbo First of all, this project is really good. We ran the functional simulation on ModelSim and worked perfectly however when we ran the timing simulation-after the synthesis on Quartus lite-it kept giving us a RESULT"1111" for a while and no output was given after that. Here is a link with the timing simulation

ZFTurbo, 2022-01-12:
Looks like we solved it. Sorry for long response:
https://github.com/ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA/issues/21