ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA #20: some question about verilog/code/neuroset

kangliyu1, 2022-01-10:
Hello! I am learning the verilog code of neural network accelerator recently. I saw your project. I am very interested in your verilog code of neural network accelerator generated using python. I admire the work you have done, so I want to spend time correcting The neuroset code you generated is simulated and verified, but there are some problems during the simulation, so I would like to ask you for advice:
1. I am using convert_image_for_testbench.py to convert the image into storage data, and replace it with the storage data in the testbench. I have tried a number of different numbers but the final simulation results are the same. The output is interrupted at 47349us "a "(as shown in Figure 1), I am very puzzled about this, I hope the seniors can give me some pointers, thank you very much! !
Note: The code used is the code in verilog/code/neuroset/
fig1:
[image]

ZFTurbo, 2022-01-10:
After using `convert_image_for_testbench.py` did you replace image in testbench.v file? Can you upload your version of testbench.v?

ZFTurbo, 2022-01-10:
`num_conv` variable in `r05_verilog_generator_neural_net_structure.py` file only regulate how many computational blocks will be generated in verilog code. Structure of neural net in this project always the same.

kangliyu1, 2022-01-11:
> After using `convert_image_for_testbench.py` did you replace image in testbench.v file? Can you upload your version of testbench.v?
@ZFTurbo Yes, I replaced the storage part of the initial in the testbench
testbench.v
```
module test();
parameter SIZE=12;
  reg clk;
  reg GO;
  reg signed [SIZE-1:0] storage [0:783]; 
  
  reg we_database;
  reg [SIZE-1:0] dp_database;
  reg [12:0] address_p_database;
  
  reg [9:0] x;
  wire STOP;
  wire [3:0] RESULT;
  TOP TOP(
	.clk					(clk),
	.GO						(GO),
	.RESULT					(RESULT),
	.we_database			(we_database), 
	.dp_database			(dp_database), 
	.address_p_database		(address_p_database-1'b1),
	.STOP					(STOP)
  );
initial begin
  clk=0;
  address_p_database=0;
  x=0;
  we_database=1;
  #200 GO=1;
end
always #100 clk=~clk;
always @(posedge clk)
	begin
		if (we_database)
		begin
			if (address_p_database<=783) 
				begin
						dp_database = storage[address_p_database];
						address_p_database=address_p_database+1'b1;
				end
			else we_database=0;
		end
		if ((x<=28*28)&&(GO)) x=x+1;
		else GO=0;
	if (STOP==1)
	begin
		$display("RESULT: %d",RESULT);
		$finish;
	end
 end
 
 
 // Precision: 12
// Image size: 28x28
// Answer: 1
initial
begin
	storage[0] =  11'b10001010000; // [0.5390625]
	storage[1] =  11'b10001110000; // [0.5546875]
	storage[2] =  11'b10010010000; // [0.5703125]
	storage[3] =  11'b10010101000; // [0.58203125]
	storage[4] =  11'b10010110000; // [0.5859375]
	storage[5] =  11'b10011000000; // [0.59375]
	storage[6] =  11'b10011011000; // [0.60546875]
	storage[7] =  11'b10011100000; // [0.609375]
	storage[8] =  11'b10011101000; // [0.61328125]
	storage[9] =  11'b10011110000; // [0.6171875]
[... 752 lines cut ...]
	storage[762] =  11'b10011111000; // [0.62109375]
	storage[763] =  11'b10100000000; // [0.625]
	storage[764] =  11'b10100001000; // [0.62890625]
	storage[765] =  11'b10100010000; // [0.6328125]
	storage[766] =  11'b10100100000; // [0.640625]
	storage[767] =  11'b10100100000; // [0.640625]
	storage[768] =  11'b10011001000; // [0.59765625]
	storage[769] =  11'b01101000000; // [0.40625]
	storage[770] =  11'b01101100000; // [0.421875]
	storage[771] =  11'b10000011000; // [0.51171875]
	storage[772] =  11'b10001101000; // [0.55078125]
	storage[773] =  11'b01111010000; // [0.4765625]
	storage[774] =  11'b01111101000; // [0.48828125]
	storage[775] =  11'b10001111000; // [0.55859375]
	storage[776] =  11'b10010100000; // [0.578125]
	storage[777] =  11'b10011000000; // [0.59375]
	storage[778] =  11'b10011101000; // [0.61328125]
	storage[779] =  11'b10011101000; // [0.61328125]
	storage[780] =  11'b10011011000; // [0.60546875]
	storage[781] =  11'b10011001000; // [0.59765625]
	storage[782] =  11'b10010111000; // [0.58984375]
	storage[783] =  11'b10010110000; // [0.5859375]
end
endmodule
```

kangliyu1, 2022-01-11:
> `num_conv` variable in `r05_verilog_generator_neural_net_structure.py` file only regulate how many computational blocks will be generated in verilog code. Structure of neural net in this project always the same.

thank you for your reply! ! ! What do you mean by computational blocks here? Is it a DSP-like resource? The more computational blocks, the faster the calculation, right? In the current verilog/code/neuroset verilog, it seems that the neural network architecture of Figure 1 cannot be seen. For example, maxpooling is only called once in TOP, but maxpooling is used 3 times in the architecture of Figure 1.
fig:1
[image]
fig:2
[image]

kangliyu1, 2022-01-11:
> `num_conv` variable in `r05_verilog_generator_neural_net_structure.py` file only regulate how many computational blocks will be generated in verilog code. Structure of neural net in this project always the same.

@ZFTurbo Hello! After reading your paper, I still have a few questions I want to ask you
1. I did not see the comparison of the hardware and software implementation results of the neural network architecture in your paper (such as the comparison of speed), I am very curious about this
2. What is the maximum value that num_conv can give when FPGA resources are sufficient? If num_conv gives a large amount, how much will the speed increase?
 Hope you can give some pointers, thank you very much! !

closed 2022-02-11 by kangliyu1

ECEVLSIWorld, 2022-12-18:
> > After using `convert_image_for_testbench.py` did you replace image in testbench.v file? Can you upload your version of testbench.v?
> > @ZFTurbo Yes, I replaced the storage part of the initial in the testbench
> > For example, the following display "// Answer: 1", the result should be 1, but the result is 2
> > However, after the storage data generated in the following figure 1 is simulated, the all result is equal to "a"
> > And so far no simulation correct image data has been found in the data generated by convert_image_for_testbench.py. I'm very confused about this, I don't know where the problem is, I hope you can give pointers, thank you very much
> > fig:1
> > [image]
> 
> testbench.v
> 
> ```
> module test();
> parameter SIZE=12;
>   reg clk;
>   reg GO;
>   reg signed [SIZE-1:0] storage [0:783]; 
>   
>   reg we_database;
>   reg [SIZE-1:0] dp_database;
>   reg [12:0] address_p_database;
>   
>   reg [9:0] x;
>   wire STOP;
>   wire [3:0] RESULT;
>   TOP TOP(
> 	.clk					(clk),
> 	.GO						(GO),
> 	.RESULT					(RESULT),
> 	.we_database			(we_database), 
> 	.dp_database			(dp_database), 
> 	.address_p_database		(address_p_database-1'b1),
> 	.STOP					(STOP)
>   );
> initial begin
>   clk=0;
>   address_p_database=0;
>   x=0;
>   we_database=1;
>   #200 GO=1;
> end
> always #10  clk=~clk;
> always @(posedge clk)
> 	begin
> 		if (we_database)
> 		begin
> 			if (address_p_database<=783) 
> 				begin
> 						dp_database = storage[address_p_database];
> 						address_p_database=address_p_database+1'b1;
> 				end
> 			else we_database=0;
> 		end
> 		if ((x<=28*28)&&(GO)) x=x+1;
> 		else GO=0;
> 	if (STOP==1)
> 	begin
> 		$display("RESULT: %d",RESULT);
> 		$finish;
> 	end
>  end
>  
>  
>  // Precision: 12
> // Image size: 28x28
> // Answer: 1
> initial
> begin
> 	storage[0] =  11'b10001010000; // [0.5390625]
> 	storage[1] =  11'b10001110000; // [0.5546875]
> 	storage[2] =  11'b10010010000; // [0.5703125]
[... 763 lines cut ...]
> 	storage[766] =  11'b10100100000; // [0.640625]
> 	storage[767] =  11'b10100100000; // [0.640625]
> 	storage[768] =  11'b10011001000; // [0.59765625]
> 	storage[769] =  11'b01101000000; // [0.40625]
> 	storage[770] =  11'b01101100000; // [0.421875]
> 	storage[771] =  11'b10000011000; // [0.51171875]
> 	storage[772] =  11'b10001101000; // [0.55078125]
> 	storage[773] =  11'b01111010000; // [0.4765625]
> 	storage[774] =  11'b01111101000; // [0.48828125]
> 	storage[775] =  11'b10001111000; // [0.55859375]
> 	storage[776] =  11'b10010100000; // [0.578125]
> 	storage[777] =  11'b10011000000; // [0.59375]
> 	storage[778] =  11'b10011101000; // [0.61328125]
> 	storage[779] =  11'b10011101000; // [0.61328125]
> 	storage[780] =  11'b10011011000; // [0.60546875]
> 	storage[781] =  11'b10011001000; // [0.59765625]
> 	storage[782] =  11'b10010111000; // [0.58984375]
> 	storage[783] =  11'b10010110000; // [0.5859375]
> end
> endmodule
> ```
> 
> [image]
Dear sir,
         Thanks for your code ,it is really helpful for me ,but I have a doubt in convert_image_for_testbench.py codes I  have to give test image number as 2 ,but default it is 0 ,i don't have idea  to change the code for giving test image as 2 , could you please help me to solve this problem.

ECEVLSIWorld, 2022-12-18:
Dear sir,
         Thanks for your code ,it is really helpful for me ,but I have a doubt in convert_image_for_testbench.py codes I  have to give test image number as 2 ,but default it is 0 ,i don't have idea  to change the code for giving test image as 2 , could you please help me to solve this problem.