ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA #21: On the simulation problem of neuroset's verilog

kangliyu1, 2022-01-12:
Hello! In the process of using your project, I encountered some problems and would like to ask you:
1. I am using convert_image_for_testbench.py ​​to generate the storage data as shown in Figure 1, and replace the storage data in the testbench, but the final simulation results of these data are all "a", and I have not found a correct simulation. storage data, I am very puzzled about this, I hope you can answer it, Figure 2 is the code structure of my project.，Figure 3 is the simulation waveform

Figure 1：
[image]
Figure 2：
[image]
Figure 3：
[image]

testbench:
```
module test();
parameter SIZE=12;
  reg clk;
  reg GO;
  reg signed [SIZE-1:0] storage [0:783]; 
  reg we_database;
  reg [SIZE-1:0] dp_database;
  reg [12:0] address_p_database;  
  reg [9:0] x;
  
  wire [3:0] RESULT;
  TOP TOP(
	.clk					(clk),
	.GO						(GO),
	.RESULT					(RESULT),
	.we_database			(we_database), 
	.dp_database			(dp_database), 
	.address_p_database		(address_p_database-1'b1),
	.STOP					(STOP)
  );
initial begin
  clk=0;
  address_p_database=0;
  x=0;
  we_database=1;
  #200 GO=1;
end
always #10 clk=~clk;
always @(posedge clk)
	begin
		if (we_database)
		begin
			if (address_p_database<=783) 
				begin
						dp_database = storage[address_p_database];
						address_p_database=address_p_database+1'b1;
				end
			else we_database=0;
		end
		if ((x<=28*28)&&(GO)) x=x+1;
		else GO=0;
	if (STOP==1)
	begin
		$display("RESULT: %d",RESULT);
		$finish;
	end
 end
 
 // Precision: 12
// Image size: 28x28
// Answer: 5

initial
begin
	storage[0] =  11'b10001010000; // [ 0.5390625]
	storage[1] =  11'b10001100000; // [ 0.546875]
	storage[2] =  11'b10001110000; // [ 0.5546875]
	storage[3] =  11'b10010000000; // [ 0.5625]
[... 758 lines cut ...]
	storage[762] =  11'b10010100000; // [ 0.578125]
	storage[763] =  11'b10010110000; // [ 0.5859375]
	storage[764] =  11'b10010110000; // [ 0.5859375]
	storage[765] =  11'b10010111000; // [ 0.58984375]
	storage[766] =  11'b10010110000; // [ 0.5859375]
	storage[767] =  11'b10001101000; // [ 0.55078125]
	storage[768] =  11'b01111011000; // [ 0.48046875]
	storage[769] =  11'b01111000000; // [ 0.46875]
	storage[770] =  11'b10000010000; // [ 0.5078125]
	storage[771] =  11'b10001000000; // [ 0.53125]
	storage[772] =  11'b10000100000; // [ 0.515625]
	storage[773] =  11'b01111001000; // [ 0.47265625]
	storage[774] =  11'b01101001000; // [ 0.41015625]
	storage[775] =  11'b01101100000; // [ 0.421875]
	storage[776] =  11'b10001000000; // [ 0.53125]
	storage[777] =  11'b10010001000; // [ 0.56640625]
	storage[778] =  11'b10010000000; // [ 0.5625]
	storage[779] =  11'b10010000000; // [ 0.5625]
	storage[780] =  11'b10001110000; // [ 0.5546875]
	storage[781] =  11'b10001100000; // [ 0.546875]
	storage[782] =  11'b10001010000; // [ 0.5390625]
	storage[783] =  11'b10001000000; // [ 0.53125]
end
 endmodule
```

ZFTurbo, 2022-01-12:
Looks like we found a problem in testbench and image_converter. It must be 1 bit lower. We made a change in repo. Now it must work ok.

kangliyu1, 2022-01-12:
> Looks like we found a problem in testbench and image_converter. It must be 1 bit lower. We made a change in repo. Now it must work ok.
> 
> Can you please check at your side?

Thank you very much for your reply. I am not in the laboratory now. I will try it tomorrow morning. Thank you very much for your patience and care. I admire you very much.

kangliyu1, 2022-01-13:
> Looks like we found a problem in testbench and image_converter. It must be 1 bit lower. We made a change in repo. Now it must work ok.
> 
> Can you please check at your side?

Hello! After your modification, are your simulation results correct? I only recognize "0" here, and the others are still all recognized as "a", I hope you can give me some pointers, thank you very much!

closed 2022-02-11 by kangliyu1