ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA #22: Some questions about network architecture

kangliyu1, 2022-01-12:
Hello! In the process of using your project, I encountered some problems and would like to ask you:
1. In the current verilog/code/neuroset verilog, the neural network architecture of Figure 1 does not seem to be seen. For example, maxpooling is called only once in TOP, but maxpooling is used 3 times in the architecture of Figure 1, as if the code and architecture diagram do not correspond
2. After reading your paper, I did not find a comparison of the software and hardware implementation results of your neural network architecture (such as speed comparison, etc.). After running python r01_train_neural_net_and_prepare_initial_weights.py, I found that the accuracy rate: 84.97%, as shown in Figure 2,but the code marked Accuracy : 97.39%, as shown in Figure 3. Is there any way to improve Accuracy?
Can the maximum value given by num_conv only be 4 when the FPGA resources are sufficient?

fig1:
[image]
fig2:
[image]
fig3:
[image]

MaKaRoIIIKa, 2022-01-12:
@kangliyu1 Hello! You are wrong in your first question. In the TOP.v file, you can see that the maxpooling layer occurs more than once (lines 183, 211, 217). The architecture inherent in the hardware implementation is fully consistent with Figure 1.

ZFTurbo, 2022-01-12:
`Can the maximum value given by num_conv only be 4 when the FPGA resources are sufficient?`

We tested only 1, 2 and 4. It's of course possible to use 8 or 16, but probably will require changes in verilog.

ZFTurbo, 2022-01-12:
`I did not find a comparison of the software and hardware implementation results of your neural network architecture (such as speed comparison, etc.)`

We didn't do such comparsion. I think GPU will beat FPGA easily in this case )

ZFTurbo, 2022-01-12:
`After running python r01_train_neural_net_and_prepare_initial_weights.py, I found that the accuracy rate: 84.97%, as shown in Figure 2,but the code marked Accuracy : 97.39%, as shown in Figure 3. Is there any way to improve Accuracy?`

Can you please add more details how you get these numbers?

kangliyu1, 2022-01-12:
> @kangliyu1 Hello! You are wrong in your first question. In the TOP.v file, you can see that the maxpooling layer occurs more than once (lines 183, 211, 217). The architecture inherent in the hardware implementation is fully consistent with Figure 1.

Thanks for your reply, I really didn't realize it was using maxp_en = 1 to enable this whole layer, I'll be learning about it tomorrow, thank you

kangliyu1, 2022-01-12:
> num_conv

Thank you for your reply. If num_conv is increased, where should verilog be modified? Why do you need to modify verilog when you change to 1, 2, and 4, but you need to modify verilog if you continue to modify num_conv? Can you give more detailed advice? Hope to get your advice, thank you very much! !

kangliyu1, 2022-01-12:
> `Can the maximum value given by num_conv only be 4 when the FPGA resources are sufficient?`
> 
> We tested only 1, 2 and 4. It's of course possible to use 8 or 16, but probably will require changes in verilog.

@ZFTurbo Thank you for your reply. If num_conv is increased, where should verilog be modified? Why do you need to modify verilog when you change to 1, 2, and 4, but you need to modify verilog if you continue to modify num_conv? Can you give more detailed advice? Hope to get your advice, thank you very much! !

kangliyu1, 2022-01-12:
> We didn't do such comparsion. I think GPU will beat FPGA easily in this case )

If the GPU can easily beat the FPGA, then the point of your work is to reduce power consumption and cost, right?

kangliyu1, 2022-01-12:
> `After running python r01_train_neural_net_and_prepare_initial_weights.py, I found that the accuracy rate: 84.97%, as shown in Figure 2,but the code marked Accuracy : 97.39%, as shown in Figure 3. Is there any way to improve Accuracy?`
> 
> Can you please add more details how you get these numbers?

I just ran "python r01_train_neural_net_and_prepare_initial_weights.py" and printed out "Accuracy: 84.97%" in the terminal, as shown in Figure 1, and then "Accuracy: 97.39%" This data is the content of the comments in the code as shown in Figure 2 Show
fig1:
[image]
fig2:
[image]

ZFTurbo, 2022-01-13:
I don't remember exactly where 97 came from. But most likely 97% accuracy was on training data which was different from what we have from camera. Images in test folder was taken from camera, and they differ from training data, so accuracy was reduced.

Also I see in log:
```
Skip training and load weights from weights\keras_model_low_weights_digit_detector.h5
```

You can delete file `weights\keras_model_low_weights_digit_detector.h5` then training will start from begining.

kangliyu1, 2022-01-28:
> @kangliyu1 Hello! You are wrong in your first question. In the TOP.v file, you can see that the maxpooling layer occurs more than once (lines 183, 211, 217). The architecture inherent in the hardware implementation is fully consistent with Figure 1.

@MaKaRoIIIKa thank you for your reply! ! Are you in charge of the verilog part of this project? I would like to ask if you have code comments for the verilog part (such as comments about the port signal function) or other information. If you have it, you can share it, thank you very much! !

closed 2022-02-11 by kangliyu1