ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA #25: some questions about project

kangliyu1, 2022-01-25:
Hello! Thank you very much for your previous reply. In the past few days, I still have some doubts about your project. I would like to ask you:
1. When using the latest project to run python r04_verilog_generator_grayscale_file.py, the error shown in Figure 1 occurs. The guess is that the parameters are missing, so I changed "get_optimal_bit_for_weights()" to "get_optimal_bit_for_weights(parse_opt())", I don't know Whether this modification is correct, so I would like to ask you, I hope you can give me some pointers.
2. After running python r05_verilog_generator_neural_net_structure.py after modifying it according to the first question, the error shown in Figure 2 appears. This error seems to mean that the index is out of range, but I don't know where to modify it, so I still want to ask you
3. After modifying the name of the file keras_model_low_weights_digit_detector.h5 according to your instructions to retrain it, it was found that an error occurred when the training reached 156/2000, which caused the interruption, as shown in Figure 3

fig1：
[image]
fig2:
[image]

fig3:
[image]

ZFTurbo, 2022-01-25:
1) Yes, sorry, I forgot to make changes in other files which used function `get_optimal_bit_for_weights`. I made a change that it now works fine.
2) Looks like in latest tensorflow changed how they define input layers input shape. I also made a fix. Now it works ok.
3) I don't know what can be the problem with training... Error appears in the middle of training, very strange.

kangliyu1, 2022-01-25:
> 1. Yes, sorry, I forgot to make changes in other files which used function `get_optimal_bit_for_weights`. I made a change that it now works fine.
> 2. Looks like in latest tensorflow changed how they [...]

Thank you for your reply, it can work correctly now, by the way, I would like to ask: your other open source project (MobileNet-in-FPGA), can this project work normally? Do you have any suggestions if I want to learn verilog for this neural network accelerator? For example: how to simulate like this project?
Thank you so much for your patience
Best wishes to you!!

ZFTurbo, 2022-01-26:
MobileNet it was our next project after this small simple net. If you download ZIP-file with project:
https://github.com/ZFTurbo/MobileNet-in-FPGA/releases/download/v2.0/OpenVino_MobileNet_verilog_project_people.zip

There is `testbench.v` in `MobileNet_v3_conv_8_3x1` folder, so you can simulate how neural net is works. But MobileNet is much larger than current project so it can be more difficult.

closed 2022-02-11 by kangliyu1