ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA #26: Some problems with reading verilog code

kangliyu1, 2022-02-11:
Hello, I found it difficult to understand some of the details in the process of reading the verilog code. Do you have any suggestions? Do you have information on the role of each module's signal ports? (For example, the role of each signal port of maxp, the role of the border module, etc.) I hope you can give me some pointers, thank you very much! !

MaKaRoIIIKa, 2022-02-14:
Hello, @kangliyu1 ! I am preparing a document for you with module pin assignments. It will take some time, you will have to wait.

kangliyu1, 2022-02-14:
> Hello, @kangliyu1 ! I am preparing a document for you with module pin assignments. It will take some time, you will have to wait.

Thank you so much! ! !
Thank you so much! ! !
Thank you so much! ! !

ZFTurbo, 2022-02-20:
We prepeared documentation with description of each pin for every module in project. Feel free to ask any question.

https://github.com/ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA/blob/master/docs/modules_pins_eng.docx

kangliyu1, 2022-02-22:
> We prepeared documentation with description of each pin for every module in project. Feel free to ask any question.
> 
> https://github.com/ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA/blob/master/docs/modules_pins_eng.docx

You helped me a lot, thank you so much

closed 2022-02-22 by kangliyu1