ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA #27: some questions about storage data

kangliyu1, 2022-02-24:
Hello! I have a question for you to ask:
Is there a way to change the storage data in the database into unsigned data?

ZFTurbo, 2022-02-24:
Why you need this? We use signed integers.

kangliyu1, 2022-02-24:
> Why you need this? We use signed integers.

Due to the project requirements, I divided the database module to the outside of the TOP, and planned to input it into the top module externally. The top module was mapped to the FPGA, and the database module was in the test circuit outside the FPGA. Therefore, there is a test circuit between the TOP module and the database module. Signed number data transmission, but the two modules are connected through the IO port of the FPGA, so the signed number cannot be directly transmitted, the symbol will be lost, so I want to ask how to convert it to an unsigned number, or if I Does doing sign bit extension make a difference?