ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA #33: convert_image_for_testbench.py

ECEVLSIWorld, 2022-12-18:
Dear sir,
         Thanks for your code ,it is really helpful for me ,but I have a doubt in convert_image_for_testbench.py codes I  have to give test image number as 2 ,but default it is 0 ,i don't have idea  to change the code for giving test image as 2 , could you please help me to solve this problem.

ECEVLSIWorld, 2022-12-19:
Could any one answer my question

ZFTurbo, 2022-12-19:
Can you please write mode deatils? We couldn't understand wht the problem is.

ECEVLSIWorld, 2022-12-19:
> <ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA/issues/33/1357305341
> @github.com>

Sir ,i have  to generate  image 2 as test bench value to top module,so in
convert image to test bench .py file consists only for image 0
conversation,but I need image 2 ,so which Line in the code should be
modified to get the test image 2  values

ECEVLSIWorld, 2022-12-19:
Please reply soon sir

ZFTurbo, 2022-12-19:
Run converter with `python convert_image_for_testbench.py --use_image 2`

ECEVLSIWorld, 2022-12-19:
> <ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA/issues/33/1357315804

It's not changing sir ,i have tried, could you please mention line number

ZFTurbo, 2022-12-19:
https://github.com/ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA/blob/master/utils/convert_image_for_testbench.py#L127

You can add after line 127:
`use_image = 2`

MaKaRoIIIKa, 2022-12-19:
https://drive.google.com/file/d/1dBpfz9BrU0x2I-VMU-TdY8-5aVhH8tLU/view?usp=share_link

ECEVLSIWorld, 2022-12-19:
> <ZFTurbo/Verilog-Generator-of-Neural-Net-Digit-Detector-for-FPGA/issues/33/1357447124

Thanks a lot. Sir

closed 2022-12-19 by ECEVLSIWorld